bettr-xyz/tracesigma-firmware #88: [Publicity] Teaser video

yongkimleng, 2020-08-01:


zhenwei2972, 2020-08-02:
Will be shooting a quick sample of francis's UI overhaul from https://github.com/fustilio/firmware_alpha/tree/ui-overhaul, since it has not been merged to master yet, it looks great btw! 
as discussed with Ji-Haw, I only have one m5stickC device, android and iOS phones, so only able to demonstrate it performing with these devices, I do not have another m5stickC to show it interoperating as such. 
Also will be relying on the sample statistics page to explain the usage of the device, as the values are still dummy values, will show exchange value incrementing on the TraceTogether app instead. Hope that this video may be sufficient to demonstrate the use case of the TraceSigma app on the m5stickc.

jhfoo, 2020-08-02:
Zhenwei, you can pick up another M5StickC from me anytime.

zhenwei2972, 2020-08-03:
@jhfoo  Perhaps I can pop by to grab one later today? also, don't think we have an real-time stats page to show the trace sigma device doing real time interorp, it currently only shows dummy values. these device does indeed exchange with android devices and for purposes of demo video currently I think it's ok, just want to take note that there is no updated display currently, have done a quick shoot with my phone for now, perhaps as a baseline, can let me know the feedback, then I'll re-do it with subtitles.

zhenwei2972, 2020-08-03:
Draft video